SpinalHDL/SaxonSoc #2: Generation of a C header with I/O addresses

lawrie, 2019-05-16:
The memory bus addresses and offsets used by each peripheral are part of the definition of SaxonSoc or a CustomSoc, but are also needed in the C programs that drive those peripherals. It would be useful if a C header file containing those addresses and offsets could be generated.

Dolu1990, 2019-05-20:
For this one, we first need be sure the if we have to try to have an Vivado/QSys like interconnect system, which can eat everything. Or if we should be more restricting, and having the user to for instance instanciate various bridges as for instance BmbToApb3

@typingArtist

lawrie, 2019-05-24:
The more restricted system with just the Bmb and Apb3 buses is fine for my purposes.

Dolu1990, 2020-02-11:
done

closed 2020-02-11 by Dolu1990