SpinalHDL/SaxonSoc #3: Reusing FPGA pins for GPIO and other peripherals

lawrie, 2019-05-16:
For microcontrollers and systems like Arduino, it is common for the same pins to be used for GPIO and one or more other peripherals. For the output pins, a mux is typically used to switch the pins between use by the GPIO and the other peripherals. 

It would be useful if, as part of the definition of the input/output ports of a peripheral, that a mapping onto GPIO pins could be specified along with the muxes to switch the pins,

As the GPIO peripheral uses a tri-state array, the C driving code for a peripheral would need to set the pins used by the peripheral to input or output as appropriate (using the GPIO writeEnable register), and to sets bits in a register to switch the muxes. It would be useful if a C header file were generated that gave the mapping between the peripheral pins and the GPIO pins and specified the mux register addresses and bit offsets..

Dolu1990, 2019-05-20:
I made a little experiment, and one thing now i'm not sure about is how to handle the selection signal.
Is that per IO, or maybe for some things, like SPI, you have a single selector which will then switch multiple pins to the SPI peripheral ?

Dolu1990, 2019-05-20:
Also, is it common to be able to have for example : 
One SPI controller which can have it's SPI pins set on multiple location, let's say you can either set them on GPIOA or GPIOB

Should the direction of the pins set by the selected functionality ? or it should always be set by the GPIO controller ?

lawrie, 2019-05-21:
For SPI, I currently have one selection signal that switches all the pins. But for some other peripherals it would be better per I/O. For example I have PWM and Servo peripherals which have a width parameters  and for these it would be better to be able to switch each IO separately.

I don't currently have the ability to do the multiple locations. I don't think that is really necessary. The setup I am testing with MuraxArduino has two SPI peripherals; one of GPIOA and one on GPIOB.

Currently I set the direction of the pins with the GPIO controller. I think that is probably OK.

roman3017, 2020-02-17:
Maybe having it consistent with Linux could be useful:
https://www.kernel.org/doc/Documentation/pinctrl.txt