SpinalHDL/SaxonSoc #5: Problem with SdramModel when changing sdram layout

roman3017, 2019-07-07:
There seems to be a problem with SdramModel. In particular, one can run De1SocLinuxSystemSim without any issues with:
```sh
sbt "runMain saxon.board.terasic.De1SocLinuxSystemSim"
```
However, if one tries to change existing sdram layout from IS42x320D to either MT48LC16M16A2 or W9825G6JH6 one gets this exception:
```sh
[info] WAITING FOR TCP JTAG CONNECTION
[info] SDRAM : MODE REGISTER DEFINITION CAS=3 burstLength=0
[info] *** VexRiscv BIOS ***
[info] *** Supervisor ***
[info] Unknown exception 00000005
[info] machineModeSbi exception
```
To reproduce, please try this patch:
```diff
diff --git a/hardware/scala/saxon/board/terasic/De1SocLinux.scala b/hardware/scala/saxon/board/terasic/De1SocLinux.scala
index 42a7da0..81f5fc3 100644
--- a/hardware/scala/saxon/board/terasic/De1SocLinux.scala
+++ b/hardware/scala/saxon/board/terasic/De1SocLinux.scala
@@ -8,7 +8,7 @@ import spinal.lib.com.uart.UartCtrlMemoryMappedConfig
 import spinal.lib.com.uart.sim.{UartDecoder, UartEncoder}
 import spinal.lib.generator._
 import spinal.lib.io.{Gpio, InOutWrapper}
-import spinal.lib.memory.sdram.IS42x320D
+import spinal.lib.memory.sdram.MT48LC16M16A2
 import spinal.lib.memory.sdram.sim.SdramModel
 
 
@@ -70,8 +70,8 @@ object De1SocLinuxSystem{
     cpu.config.load(VexRiscvConfigs.linux)
     cpu.enableJtag(clockCtrl)
 
-    sdramA.layout.load(IS42x320D.layout)
-    sdramA.timings.load(IS42x320D.timingGrade7)
+    sdramA.layout.load(MT48LC16M16A2.layout)
+    sdramA.timings.load(MT48LC16M16A2.timingGrade7)
 
     uartA.parameter load UartCtrlMemoryMappedConfig(
       baudrate = 115200,
```
Please let me know if there are any additional changes required when changing sdram layout.

Dolu1990, 2019-07-07:
Did you changed the linux DTS to reflect the SDRAM capacity ? 

``` scala
println(s"W9825G6JH6 => ${W9825G6JH6.layout.capacity/1024/1024} MiB")
println(s"MT48LC16M16A2 => ${MT48LC16M16A2.layout.capacity/1024/1024} MiB")
println(s"IS42x320D => ${IS42x320D.layout.capacity/1024/1024} MiB")
=> 
W9825G6JH6 => 32 MiB
MT48LC16M16A2 => 32 MiB
IS42x320D => 64 MiB
```

It is likely the issue.

roman3017, 2019-07-07:
@Dolu1990 the suggested change:
```diff
    memory@80400000 {
        device_type = "memory";
-       reg = <0x80400000 0x03C00000>;
+       reg = <0x80400000 0x01C00000>;
    };
```
indeed fixed the reported exception. The simulation passes the original problem and gets kernel panic later out of memory, which can be hopefully solved by further DTS changes. We can close this.
Thank you very much for your help.

closed 2019-07-07 by roman3017

roman3017, 2019-07-07:
Just for the record, the following DTS patch fixed the out of memory panic mentioned earlier:
```diff
    chosen {
        bootargs = "rootwait console=hvc0 root=/dev/ram0 init=/sbin/init swiotlb=32"; // loglevel=7
        linux,initrd-start = <0x80C00000>;
-       linux,initrd-end =   <0x82000000>; 
+       linux,initrd-end =   <0x81000000>; 
    };
```

Dolu1990, 2019-07-07:
@roman3017  Are you interrested into Saxon ? There is so many things to do XD So if you want to contribut, you are welcome ^^
Also, i'm currently working on a new SDRAM controller (SDR, DDRx)

roman3017, 2019-07-07:
@Dolu1990: I am interested although sometime I am too busy with my daily tasks. I am working on making Saxon to work on Scarab (MiniSpartan6+) HW, which is why I needed this change. This is the only FPGA board with SDRAM I have. Once you have DDR controller, things will be easier for me, since I can use Zybo board with DDR2 or Nexys4DDR with DDR3. I will contribute what I have and started to follow your Gitter conversation.

Dolu1990, 2019-07-08:
Ok :D
Let's me know how it goes.

I just moved the SDRAM controller from the spinal.lib.memory.sdram package to the spinal.lib.memory.sdram.sdr package.