SpinalHDL/SaxonSoc #29: Support for Arty A7 100t

vj-kumar, 2020-02-25:
Can we have support for synthesising necessary files for the 100t variant of digilent arty kit?

Dolu1990, 2020-03-18:
I'm not experienced enough in vivado tcl, seem like it would mainly be about https://github.com/SpinalHDL/SaxonSoc/blob/dev/hardware/synthesis/arty_a7/vivado_params.tcl#L2

@sebastien-riou Do you know a clean way of handeling such cases ?

sebastien-riou, 2020-03-18:
should not be a big deal indeed. there is also [https://github.com/SpinalHDL/SaxonSoc/blob/a89053c637cdbcb2c4ce93acae9dd32e429d8d21/hardware/synthesis/arty_a7/vivado_script.tcl#L16](https://github.com/SpinalHDL/SaxonSoc/blob/a89053c637cdbcb2c4ce93acae9dd32e429d8d21/hardware/synthesis/arty_a7/vivado_script.tcl#L16) to change.
at some point we should have an argument to the make file I guess

vj-kumar, 2020-03-18:
I am also not very experienced in that. Below is the diff I used to test on a 100T variant. If someone is willing to send a patch based on these parts please feel free to send it. I was planning to have a look into it, but currently occupied and it might take some time.
```diff
diff --git a/hardware/synthesis/arty_a7/vivado_params.tcl b/hardware/synthesis/arty_a7/vivado_params.tcl
index 024ce2a..6a5117d 100644
--- a/hardware/synthesis/arty_a7/vivado_params.tcl
+++ b/hardware/synthesis/arty_a7/vivado_params.tcl
@@ -1,5 +1,5 @@
 set outputdir ./vivado_project
-set part "xc7a35ticsg324-1L"
+set part "xc7a100tcsg324-1"
 set base ".."
 set projectName "fpga"
 set top Arty7Linux
diff --git a/hardware/synthesis/arty_a7/vivado_script.tcl b/hardware/synthesis/arty_a7/vivado_script.tcl
index 65a7dba..fcedfad 100644
--- a/hardware/synthesis/arty_a7/vivado_script.tcl
+++ b/hardware/synthesis/arty_a7/vivado_script.tcl
@@ -13,7 +13,7 @@ variable sbi_file
 set sbi_file "../../../../software/standalone/machineModeSbi/build/machineModeSbi.bin"
 variable boot_file
 set boot_file "../../../../../u-boot/u-boot.bin"
-variable fpga_part xc7a35t
+variable fpga_part xc7a100t

 proc help {} {
        variable script_file
```

overcookedlobster, 2022-06-11:
i'm curious, was changing those scripts enough?

vj-kumar, 2022-06-11:
As far as I remember, it was enough at that time.