SpinalHDL/SaxonSoc #32: software compile fail

wuxx, 2020-04-14:
Hi
I use this command to compile the demo code
`SaxonSoc/software/standalone/blinkRgb$` make BSP=FomuMinimal
and it report compile fail,  here is the output detail。
it seem that the program is rv32i but it use the riscv64 libc.a (64-bit), so link failed.
can somebody help me figure out how to solve this problem :(
=====================
mkdir -p build//home/ubuntu/oss/SaxonSoc/software/standalone/blinkRgb/src/
riscv64-unknown-elf-gcc -c -I../../../bsp/FomuMinimal/include -I../../../bsp/FomuMinimal/app -Os -march=rv32i -mabi=ilp32 -DUSE_GP  -I../include -I../driver   -o build//home/ubuntu/oss/SaxonSoc/software/standalone/blinkRgb/src/main.o src/main.c
mkdir -p build//home/ubuntu/oss/SaxonSoc/software/standalone/common/
riscv64-unknown-elf-gcc -c -I../../../bsp/FomuMinimal/include -I../../../bsp/FomuMinimal/app -Os -march=rv32i -mabi=ilp32 -DUSE_GP  -I../include -I../driver   -o build//home/ubuntu/oss/SaxonSoc/software/standalone/common/start.o ../common/start.S
riscv64-unknown-elf-gcc -I../../../bsp/FomuMinimal/include -I../../../bsp/FomuMinimal/app -Os -march=rv32i -mabi=ilp32 -DUSE_GP  -I../include -I../driver -o build/blinkRgb.elf build//home/ubuntu/oss/SaxonSoc/software/standalone/blinkRgb/src/main.o build//home/ubuntu/oss/SaxonSoc/software/standalone/common/start.o -march=rv32i -mabi=ilp32 -lc -nostdlib -lgcc -nostartfiles -ffreestanding -Wl,-Bstatic,-T,../../../bsp/FomuMinimal/linker/default.ld,-Map,build/blinkRgb.map,--print-memory-usage
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o)
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: build/blinkRgb.elf(.text): relocation ".L5+0x0 (type R_RISCV_BRANCH)" goes out of range
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): file class ELFCLASS64 incompatible with ELFCLASS32
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
../common/standalone.mk:31: recipe for target 'build/blinkRgb.elf' failed
make: *** [build/blinkRgb.elf] Error 1
=====================

Dolu1990, 2020-04-14:
Hi,

How did you installed your riscv compilation toolchain ?

wuxx, 2020-04-14:
I compile the riscv toolchain myself, under /opt/riscv/bin/riscv64-unknown-elf-gcc
does I need re-compile the toolchain with addition config flag ?

Dolu1990, 2020-04-14:
Personnaly, i'm using the prebuild one described in https://github.com/SpinalHDL/VexRiscv#build-the-risc-v-gcc

Should be fine then ^^

wuxx, 2020-04-15:
I try this toolchain and compile is ok now, thank you very much! :)
https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-20171231-x86_64-linux-centos6.tar.gz

closed 2020-04-15 by wuxx