SpinalHDL/SaxonSoc #57: Get SaxonSoC into the SymbiFlow examples / fpga-tool-perf

mithro, 2020-11-25:
The SymbiFlow project is slowly getting pretty decent support for the Xilinx Artix 7 part and more specifically the Digilent Arty A7 board. It would be awesome to have a SaxonSoC design in the [SymbiFlow Examples](https://github.com/SymbiFlow/symbiflow-examples) repository so we can make sure the design works with the fully open source tooling (we already have a Linux + LiteX with VexRISCV design example working). I logged https://github.com/SymbiFlow/symbiflow-examples/issues/74) on the SymbiFlow Examples.

It might also good to get SaxonSoC into the [SymbiFlow fpga-tool-perf](https://github.com/SymbiFlow/fpga-tool-perf) as a test design -- that way we can track how the performance is improved over time (and how the open tools compares to options like Vivado). (Logged https://github.com/SymbiFlow/fpga-tool-perf/issues/278 for that.)

roman3017, 2020-11-26:
@mithro: I have tried symbiflow but getting this error when trying to synthesize NexysA7SmpLinux with symbiflow_synth:
```sh
make -f symbiflow.mk 
cd build_symbiflow && symbiflow_synth -t NexysA7SmpLinux -v /.../SaxonSoc/hardware/synthesis/digilent/NexysA7SmpLinux/../../../netlist/NexysA7SmpLinux.v -d artix7 -p xc7a100tcsg324-1 2>&1 > /dev/null
ERROR: Module `\Ram_1w_1rs' referenced in module `\DataCache' in cell `\ways_1_data' is not part of the design.
make: *** [symbiflow.mk:26: build_symbiflow/NexysA7SmpLinux.eblif] Error 1
```
Here is my makefile:
```sh
# cd hardware/synthesis/digilent/NexysA7SmpLinux/
# export INSTALL_DIR=/opt/symbiflow
# export PATH="$INSTALL_DIR/xc7/install/bin:$PATH"
# source "$INSTALL_DIR/xc7/conda/etc/profile.d/conda.sh"
# conda activate xc7
# make -f symbiflow.mk
# conda deactivate

mkfile_path := $(abspath $(lastword $(MAKEFILE_LIST)))
current_dir := $(patsubst %/,%,$(dir $(mkfile_path)))
TOP:=NexysA7SmpLinux
VERILOG:=${current_dir}/../../../netlist/$(TOP).v
BITSTREAM_DEVICE := artix7
SDC:=${current_dir}/NexysA7.xdc
BUILDDIR:=build_symbiflow

PARTNAME:= xc7a100tcsg324-1
PCF:=${current_dir}/arty.pcf
DEVICE:= xc7a100t_test

all: ${BUILDDIR}/${TOP}.bit

${BUILDDIR}:
	mkdir ${BUILDDIR}

${BUILDDIR}/${TOP}.eblif: | ${BUILDDIR}
	cd ${BUILDDIR} && symbiflow_synth -t ${TOP} -v ${VERILOG} -d ${BITSTREAM_DEVICE} -p ${PARTNAME} 2>&1 > /dev/null

${BUILDDIR}/${TOP}.net: ${BUILDDIR}/${TOP}.eblif
	cd ${BUILDDIR} && symbiflow_pack -e ${TOP}.eblif -d ${DEVICE} -s ${SDC} 2>&1 > /dev/null

${BUILDDIR}/${TOP}.place: ${BUILDDIR}/${TOP}.net
	cd ${BUILDDIR} && symbiflow_place -e ${TOP}.eblif -d ${DEVICE} -p ${PCF} -n ${TOP}.net -P ${PARTNAME} -s ${SDC} 2>&1 > /dev/null

${BUILDDIR}/${TOP}.route: ${BUILDDIR}/${TOP}.place
	cd ${BUILDDIR} && symbiflow_route -e ${TOP}.eblif -d ${DEVICE} -s ${SDC} 2>&1 > /dev/null

${BUILDDIR}/${TOP}.fasm: ${BUILDDIR}/${TOP}.route
	cd ${BUILDDIR} && symbiflow_write_fasm -e ${TOP}.eblif -d ${DEVICE}

${BUILDDIR}/${TOP}.bit: ${BUILDDIR}/${TOP}.fasm
	cd ${BUILDDIR} && symbiflow_write_bitstream -d ${BITSTREAM_DEVICE} -f ${TOP}.fasm -p ${PARTNAME} -b ${TOP}.bit

clean:
	rm -rf ${BUILDDIR}
```

Dolu1990, 2020-11-26:
@roman3017 

I think your issue is because your makefile do not include the dual port with byte enable blackbox : 
https://github.com/SpinalHDL/SaxonSoc/blob/06398189392ba6e783acd9aaae8665248d063056/hardware/synthesis/xilinx/common/RamXilinx.v

This should fix it :)

roman3017, 2020-11-27:
@Dolu1990: After adding RamXilinx.v:
```sh
VERILOG := ${current_dir}/../../../netlist/$(TOP).v ${current_dir}/../../xilinx/common/RamXilinx.v
```
symbiflow_synth runs for a while but crashes in the end:
```sh
...
7.39. Executing XILINX_DFFOPT pass (optimize FF control signal usage).
Optimizing FFs in NexysA7SmpLinux.
  Merging D + CE LUTs for $auto$simplemap.cc:420:simplemap_dff$57838/debug_logic.debugger.dispatcher_dataLoaded (3 -> 4)
  Merging D + CE LUTs for $auto$simplemap.cc:420:simplemap_dff$57839/debug_logic.debugger.dispatcher_headerLoaded (4 -> 5)
  Merging D + CE LUTs for $auto$simplemap.cc:420:simplemap_dff$57840/debug_logic.debugger.dispatcher_counter (2 -> 4)
  Merging D + CE LUTs for $auto$simplemap.cc:420:simplemap_dff$57841/debug_logic.debugger.dispatcher_counter (2 -> 5)
  Merging D + CE LUTs for $auto$simplemap.cc:420:simplemap_dff$57842/debug_logic.debugger.dispatcher_counter (2 -> 5)
  Merging D + CE LUTs for $auto$simplemap.cc:420:simplemap_dff$58246/sdramDomain_bmbCc_bmb_decoder.logic_rspHits_1 (1 -> 4)
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
/opt/symbiflow/xc7/install/bin/symbiflow_synth: line 102:  8527 Aborted                 (core dumped) yosys -p "tcl ${SYNTH_TCL_PATH}" -l $LOG ${VERILOG_FILES[*]}
```

Dolu1990, 2020-11-27:
@roman3017 Ahhh sadness 

that's with the upstream flow ?

roman3017, 2020-11-27:
@Dolu1990: I have used instructions from  [SymbiFlow Examples](https://github.com/SymbiFlow/symbiflow-examples) repository as suggested.

roman3017, 2020-12-19:
I have also tried the latest code: https://github.com/roman3017/symbiflow-arch-defs/commit/fe8b148474b4b104c0a24730caa6bcbbf5975406
It failed with:

```sh
8.37. Executing XILINX_DFFOPT pass (optimize FF control signal usage).
Optimizing FFs in NexysA7SmpLinux.
...
  Merging D + terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
make[3]: *** [xc/xc7/tests/soc/saxon/CMakeFiles/file_xc_xc7_tests_soc_saxon_saxon_nexysa7_artix7_100t-xc7a100t-virt-xc7a100t-test_NexysA7SmpLinux_synth.sdc.dir/build.make:89: xc/xc7/tests/soc/saxon/saxon_nexysa7/artix7_100t-xc7a100t-virt-xc7a100t-test/NexysA7SmpLinux_synth.json] Error 134
make[2]: *** [CMakeFiles/Makefile2:257219: xc/xc7/tests/soc/saxon/CMakeFiles/file_xc_xc7_tests_soc_saxon_saxon_nexysa7_artix7_100t-xc7a100t-virt-xc7a100t-test_NexysA7SmpLinux_synth.sdc.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:257304: xc/xc7/tests/soc/saxon/CMakeFiles/saxon_nexysa7_bit.dir/rule] Error 2
make: *** [Makefile:281: xc/xc7/tests/soc/saxon/CMakeFiles/saxon_nexysa7_bit.dir/rule] Error 2
```

roman3017, 2021-03-22:
Things have improved recently and at least saxon synthesis seems to work. It was tested with this patch:
https://github.com/roman3017/symbiflow-arch-defs/commit/ec86481f07dd214cb3cd4e56495f11f2aeee4ae6
However VPR is failing:
```sh
make saxon_nexysa7_bit
...
Error 1: 
Type: Blif file
File: /home/roman/projects/riscv/symbiflow-arch-defs/build/xc/xc7/tests/soc/saxon/saxon_nexysa7/artix7_100t-xc7a100t-virt-xc7a100t-test/NexysA7SmpLinux.eblif
Line: 526471
Message: Failed to find matching architecture model for 'MMCME2_ADV'

Found constant-zero generator '$abc$843313$lut$auto$rtlil.cc:2281:NotGate$249472.T0'
Found constant-zero generator '$abc$843313$lut$flatten\system_cores_0_logic_cpu.$logic_and$/home/roman/projects/riscv/symbiflow-arch-defs/build/xc/xc7/tests/soc/saxon/NexysA7SmpLinux.v:40041$22060_Y.T1'
Found constant-zero generator '$abc$843313$lut$flatten\system_cores_1_logic_cpu.$logic_and$/home/roman/projects/riscv/symbiflow-arch-defs/build/xc/xc7/tests/soc/saxon/NexysA7SmpLinux.v:30747$19491_Y.T1'
Found constant-zero generator '$abc$843313$lut$flatten\system_fpu_logic.$procmux$37049_Y.T0'
Found constant-zero generator '$abc$843313$lut\system_bmbPeripheral_bmb_decoder._zz_30.T0'
Found constant-zero generator '$abc$843313$lut\system_bmbPeripheral_bmb_decoder.logic_hits_3.T0'
Found constant-zero generator '$abc$843313$lut\system_bmbPeripheral_bmb_decoder.logic_hits_4.T0'
Found constant-zero generator '$abc$843313$lut\system_bmbPeripheral_bmb_decoder.logic_hits_5.T0'
Found constant-zero generator '$abc$843313$lut\system_cores_0_logic_cpu.MmuPlugin_ports_1_requireMmuLockupCalc.T1'
Found constant-zero generator '$abc$843313$lut\system_cores_0_logic_cpu._zz_679.T0'
Found constant-zero generator '$abc$843313$lut\system_cores_0_logic_cpu.decode_SRC_USE_SUB_LESS.T1'
Found constant-zero generator '$abc$843313$lut\system_cores_1_logic_cpu.MmuPlugin_ports_0_cacheHitsCalc[0].T1'
Found constant-zero generator '$abc$843313$lut\system_cores_1_logic_cpu.MmuPlugin_ports_0_cacheHitsCalc[2].T1'
Found constant-zero generator '$abc$843313$lut\system_cores_1_logic_cpu.MmuPlugin_ports_1_requireMmuLockupCalc.T1'
Found constant-zero generator '$abc$843313$lut\system_dma_logic.b2s_1_cmd_context_veryLast.T0'
Found constant-zero generator '$abc$843313$lut\system_fabric_dBus_bmb_decoder.portsLogic_0_rspFifo._zz_1.T0'
Found constant-zero generator '$abc$843313$lut\system_fabric_dBus_bmb_decoder.portsLogic_1_rspFifo._zz_1.T0'
Found constant-zero generator '$abc$843313$lut\system_fabric_dBus_bmb_decoder.sourceOrderingFifo._zz_1.T0'
Found constant-zero generator '$auto$xilinx_dffopt.cc:341:execute$881616.T0'
Found constant-zero generator '$auto$xilinx_dffopt.cc:341:execute$881950.T0'
# Load circuit took 2.55 seconds (max_rss 172.5 MiB, delta_rss +138.2 MiB)
The entire flow of VPR took 2.92 seconds (max_rss 172.5 MiB)
make[3]: *** [xc/xc7/tests/soc/saxon/CMakeFiles/file_xc_xc7_tests_soc_saxon_saxon_nexysa7_artix7_100t-xc7a100t-virt-xc7a100t-test_NexysA7SmpLinux.net.dir/build.make:88: xc/xc7/tests/soc/saxon/saxon_nexysa7/artix7_100t-xc7a100t-virt-xc7a100t-test/NexysA7SmpLinux.net] Error 1
make[2]: *** [CMakeFiles/Makefile2:257222: xc/xc7/tests/soc/saxon/CMakeFiles/file_xc_xc7_tests_soc_saxon_saxon_nexysa7_artix7_100t-xc7a100t-virt-xc7a100t-test_NexysA7SmpLinux.net.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:257460: xc/xc7/tests/soc/saxon/CMakeFiles/saxon_nexysa7_bit.dir/rule] Error 2
make: *** [Makefile:311: xc/xc7/tests/soc/saxon/CMakeFiles/saxon_nexysa7_bit.dir/rule] Error 2
```
I have also tried ArtyA7 target and it is also failing after synthesis:
```sh
make saxon_artya7_bit
...
Error 1: 
Type: Blif file
File: /home/roman/projects/riscv/symbiflow-arch-defs/build/xc/xc7/tests/soc/saxon/saxon_artya7/artix7-xc7a50t-virt-xc7a50t-test/ArtyA7SmpLinux.eblif
Line: 528502
Message: Failed to find matching architecture model for 'OSERDESE2'

Found constant-zero generator '$abc$843918$lut\system_cores_1_logic_cpu.decode_SRC_USE_SUB_LESS.T1'
Found constant-zero generator '$abc$843918$lut\system_dma_logic.b2s_1_cmd_context_veryLast.T0'
Found constant-zero generator '$abc$843918$lut\system_fabric_dBus_bmb_decoder.portsLogic_0_rspFifo._zz_1.T0'
Found constant-zero generator '$abc$843918$lut\system_fabric_dBus_bmb_decoder.portsLogic_1_rspFifo._zz_1.T0'
Found constant-zero generator '$abc$843918$lut\system_fabric_dBus_bmb_decoder.sourceOrderingFifo._zz_1.T0'
Found constant-zero generator '$abc$843918$lut\system_fpu_logic.roundFront_mantissaIncrement.T1'
# Load circuit took 2.50 seconds (max_rss 169.6 MiB, delta_rss +138.9 MiB)
The entire flow of VPR took 2.84 seconds (max_rss 169.6 MiB)
make[3]: *** [xc/xc7/tests/soc/saxon/CMakeFiles/file_xc_xc7_tests_soc_saxon_saxon_artya7_artix7-xc7a50t-virt-xc7a50t-test_ArtyA7SmpLinux.net.dir/build.make:88: xc/xc7/tests/soc/saxon/saxon_artya7/artix7-xc7a50t-virt-xc7a50t-test/ArtyA7SmpLinux.net] Error 1
make[2]: *** [CMakeFiles/Makefile2:257920: xc/xc7/tests/soc/saxon/CMakeFiles/file_xc_xc7_tests_soc_saxon_saxon_artya7_artix7-xc7a50t-virt-xc7a50t-test_ArtyA7SmpLinux.net.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:258106: xc/xc7/tests/soc/saxon/CMakeFiles/saxon_artya7_bit.dir/rule] Error 2
make: *** [Makefile:611: xc/xc7/tests/soc/saxon/CMakeFiles/saxon_artya7_bit.dir/rule] Error 2
```
There is also a chance that everything works now and I am just missing something.