SpinalHDL/SaxonSoc #59: ULX3S - Synthesis issue

soundnut, 2021-03-12:
Hi,
I'm trying to rebuild all components as outlined under Build everything of the readme (https://github.com/SpinalHDL/SaxonSoc/blob/dev/bsp/Ulx3sLinuxUboot/README.md)

The software parts work but I stumble while building the bitstream (make in SaxonSoc/hardware/synthesis/ulx3s) It fails consistently with this error:
 4.44.22.3. Executing XAIGER backend.
ERROR: Visited AIG node more than once; this could be a combinatorial loop that has not been broken - see Yosys bug 2530
make: *** [makefile:41: bin/toplevel.json] Error 1

So far, I've tried to completely start from scratch, double-checked all pre-requisites, made sure that all pre-requisites are on the latest build etc. 

Any Idea what I could do to rectify this?

BTW: how/where do I specify the size of the FPGA (ULX3S comes with different FPGA sizes)? 

Thanks

Dolu1990, 2021-03-12:
Hi,

Hooo dev is a old branch, can you give a try on dev-0.2 ?
https://github.com/SpinalHDL/SaxonSoc/tree/dev-0.2/bsp/radiona/ulx3s/smp

There you will also find doc for the FPGA size (FPGA_SIZE=85 saxon_bitstream) for example.

soundnut, 2021-03-12:
Hi 
Thanks for your fast answer - much appreciated.
Apologies - didn't realize that I'm fiddling with an outdated version. Will restart from scratch with dev-0.2
I'll close the issue.

closed 2021-03-12 by soundnut

soundnut, 2021-03-13:
For the record: managed to build a 2 CPU system with dev-0.2

Dolu1990, 2021-03-14:
Cool thanks :)

Dolu1990, 2021-03-14:
I'm currently working on the "fiber" branch which will improve the SoC syntax and wipe away the Generator stuff for a simpler paradigm. Will soon merge that in a new dev-0.3 branch, when ulx3s is ported on it.