SpinalHDL/SaxonSoc #60: 4 CPU fails in buildroot (dts)

soundnut, 2021-03-13:
Hi,
Managed to build a 2 CPU System fine but any attempt to build with 4 CPUs fails while generating linux.dts 

SAXON_CPU_COUNT=4 saxon_buildroot
.....
  DTC     arch/riscv/boot/dts/linux.dtb
Error: /development/Ulx3sSmp/buildroot-spinal-saxon/boards/common/dts/linux_cpu.dts:55.25-26 syntax error
FATAL ERROR: Unable to parse input tree
make[3]: *** [scripts/Makefile.lib:326: arch/riscv/boot/dts/linux.dtb] Error 1
make[2]: *** [Makefile:1338: linux.dtb] Error 2
make[1]: *** [package/pkg-generic.mk:250: /development/Ulx3sSmp/buildroot-build/build/linux-spinal-v5.10.1/.stamp_built] Error 2
make: *** [Makefile:23: _all] Error 2

Any idea how to fix this? the .dts file doesn't look like it's complete.

Thanks

Attached linux_cpu.dts 
[linux_cpu.dts.gz](https://github.com/SpinalHDL/SaxonSoc/files/6135326/linux_cpu.dts.gz)

Dolu1990, 2021-03-13:
Hooo, the issue is SAXON_CPU_ISAy  instead of SAXON_CPU_ISA

I pushed a fix : 
https://github.com/SpinalHDL/buildroot-spinal-saxon/commit/17a25f2398b494efbc921dee9184afefe2f9c16c#diff-32876606dfd7be262aa095887b04b45e4e4279b7666bde3516dd75998112ad3a

let's me know if there is any issue ^^

Dolu1990, 2021-03-13:
That DTS file is included the main one there : 
https://github.com/SpinalHDL/buildroot-spinal-saxon/blob/main/boards/spinal-saxon/ulx3s/linux/linux.dts#L24

It is just a fragment of the full config.

soundnut, 2021-03-14:
Perfect - thank you! This fixes the issue
Build completed without any further issues.

root@buildroot:~# cat /proc/cpuinfo
processor       : 0
hart            : 3
isa             : rv32ima
mmu             : sv32

processor       : 1
hart            : 0
isa             : rv32ima
mmu             : sv32

processor       : 2
hart            : 1
isa             : rv32ima
mmu             : sv32

processor       : 3
hart            : 2
isa             : rv32ima
mmu             : sv32

closed 2021-03-14 by soundnut