SpinalHDL/SaxonSoc #61: Ulx3s with Ethernet?

soundnut, 2021-03-15:
Hi again,

The running Linux image shows eth0 - is it possible to connect a  LAN8720 Microchip RMII or  ENC28J60 Adapter? if so, where can I find the correct pin-out? 

Thanks

soundnut, 2021-03-15:
This is one set of instructions I found: https://github.com/lawrie/saxonsoc-ulx3s-bin/tree/master/Smp It points back to this project at the end under "Build from Source". Tried connecting a ENC28J60 Adapter as described but can't get it to run.

Dolu1990, 2021-03-15:
I personnaly never did it, but here is the pinout for the RMII interface as a PMOD : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.2/hardware/synthesis/radiona/ulx3s/smp/ulx3s_v20_linux_uboot.lpf#L374

It works with the LAN8720 ETH board (https://www.waveshare.com/lan8720-eth-board.htm)
@lawrie used it much as far as i know.

soundnut, 2021-03-15:
ok - thanks.

I connected the LAN8720 module and configured the interface with a static IP Address. 
Good news: I can see the port on my switch and there is traffic.
Bad news: something is still wonky.

configured port mirroring so that I can monitor traffic
I can see ARP requests and outgoing ICMP requests. Incoming is sporadic 
upon closer inspection I noticed that the broadcast address is off. (check screenshot - should be .255 instead ifconfig shows .55
[image]

Related: There aren't many tools in the image - how can I add more networking tools like tcpdump to it?

Thanks

soundnut, 2021-03-15:
disregard the 2nd part - found instructions here: https://github.com/SpinalHDL/buildroot-spinal-saxon

soundnut, 2021-03-15:
interestingly, I now see dhcp queries from the device but it doesn't seem to accept the config (I added networking tools - dhcpd, among others, to the image
[image]
adding a static IP-Address and outbound ping shows ARP Requests but no ICMP (as seen from the switch port)
[image]

soundnut, 2021-03-28:
This has been solved - I have a working setup with enc28j60. Will try again with LAN8720 as soon as I receive a new one.

closed 2021-03-28 by soundnut