SpinalHDL/SaxonSoc #65: mtvec read value via debug

sebastien-riou, 2021-07-30:
Not sure if the issue is hardware or software. The first read via GDB returns a wrong value (always the same across multiple trials on FPGA). I get the correct value after a stepi (with an instruction which does not touch mtvec).

````
Reading symbols from build/Matryoshka/testRam.elf...
(gdb) set remotetimeout 10000
(gdb) set arch riscv:rv32
The target architecture is assumed to be riscv:rv32
(gdb) target extended-remote :3333
Remote debugging using :3333
0x20a00000 in ?? ()
(gdb) load
Loading section .init, size 0x94 lma 0x10a80000
Loading section .text, size 0x2006 lma 0x10a80094
Loading section .eh_frame, size 0x50 lma 0x10a8209c
Loading section .data, size 0x734 lma 0x10a820ec
Start address 0x10a80000, load size 10270
Transfer rate: 10 KB/sec, 2567 bytes/write.
(gdb) info registers $pc
pc             0x10a80000	0x10a80000 <_start>
(gdb) info registers $mtvec
mtvec          0x0	0
(gdb) continue
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x10a811b8 in satl_master_fuzzing ()
(gdb) info registers $pc   
pc             0x10a811b8	0x10a811b8 <satl_master_fuzzing+240>
(gdb) info registers $mtvec
mtvec          0x20a0114c	547361100
(gdb) info registers $mtvec
mtvec          0x20a0114c	547361100
(gdb) disassemble /r
...
   0x10a8115e <+150>:	99 a0	j	0x10a811a4 <satl_master_fuzzing+220>
   0x10a81160 <+152>:	b9 e8	bnez	s1,0x10a811b6 <satl_master_fuzzing+238>
   0x10a81162 <+154>:	73 29 00 b0	csrr	s2,mcycle
   0x10a81166 <+158>:	b3 b7 29 01	sltu	a5,s3,s2
   0x10a8116a <+162>:	33 07 39 41	sub	a4,s2,s3
...
   0x10a811b6 <+238>:	82 94	jalr	s1
=> 0x10a811b8 <+240>:	4d d5	beqz	a0,0x10a81162 <satl_master_fuzzing+154>
   0x10a811ba <+242>:	05 45	li	a0,1
   0x10a811bc <+244>:	e5 b7	j	0x10a811a4 <satl_master_fuzzing+220>
   0x10a811be <+246>:	e3 72 ea f6	bgeu	s4,a4,0x10a81122 <satl_master_fuzzing+90>
   0x10a811c2 <+250>:	65 bf	j	0x10a8117a <satl_master_fuzzing+178>
End of assembler dump.
(gdb) stepi
0x10a81162 in satl_master_fuzzing ()
(gdb) info registers $mtvec
mtvec          0x10a8084a	279447626
````

Dolu1990, 2021-08-01:
Hi,

I guess the issue is that the openocd port do not implement CSR register access. The hardware should be able to support it, as you can push instruction.

Else, another possibility is that the CPU configuration that you use do not implement reading mtvec via csr reads ?

sebastien-riou, 2021-08-01:
looks like GDB issue, openocd+telnet read it just fine, sorry I close it.
````
1 > telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* saxon.cpu0         vexriscv   little SOC_CPU.bridge     halted
 1  soc.sstap          testee     little SOC_SSTAP.bridge   unknown
 2  subsys.cpu0        vexriscv   little SUBSYS_CPU.bridge  halted
 3  subsys.sstap       testee     little SUBSYS_SSTAP.bridge unknown

> reg pc
pc (/32): 0x20A00000

> reg mtvec
mtvec (/32): 0x00000000

> resume
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* saxon.cpu0         vexriscv   little SOC_CPU.bridge     running
 1  soc.sstap          testee     little SOC_SSTAP.bridge   unknown
 2  subsys.cpu0        vexriscv   little SUBSYS_CPU.bridge  halted
 3  subsys.sstap       testee     little SUBSYS_SSTAP.bridge unknown

> reg mtvec
mtvec (/32): 0x20A01278

> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* saxon.cpu0         vexriscv   little SOC_CPU.bridge     running
 1  soc.sstap          testee     little SOC_SSTAP.bridge   unknown
 2  subsys.cpu0        vexriscv   little SUBSYS_CPU.bridge  halted
 3  subsys.sstap       testee     little SUBSYS_SSTAP.bridge unknown

accepting 'gdb' connection on tcp/3333
New GDB Connection: 1, Target saxon.cpu0, state: halted
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* saxon.cpu0         vexriscv   little SOC_CPU.bridge     halted
 1  soc.sstap          testee     little SOC_SSTAP.bridge   unknown
 2  subsys.cpu0        vexriscv   little SUBSYS_CPU.bridge  halted
 3  subsys.sstap       testee     little SUBSYS_SSTAP.bridge unknown

> reg mtvec
mtvec (/32): 0x20A01278

> 

````

GDB:
````
GNU gdb (SiFive GDB-Metal 10.1.0-2020.12.7) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/sifive/freedom-tools/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/SaxonSubsys/tiempo_bootloader.elf...
(gdb) set remotetimeout 10000
(gdb) set arch riscv:rv32
The target architecture is set to "riscv:rv32".
(gdb) target extended-remote :3333
Remote debugging using :3333
0x20a02538 in ?? ()
(gdb) info registers $pc
pc             0x20a02538	0x20a02538
(gdb) info registers $mtvec
mtvec          0x0	0
(gdb) info registers $mtvec
mtvec          0x0	0
(gdb) info registers $mtvec
mtvec          0x0	0
(gdb) stepi
0x20a0253a in ?? ()
(gdb) info registers $mtvec
mtvec          0x20a01278	547361400

````

closed 2021-08-01 by sebastien-riou