SpinalHDL/SaxonSoc #66: I want to change cpu config in Ulx3SMinimal.scala,but got error.

jmio, 2021-09-30:
Hi.

I have started to use SaxonSoC.
To change the CPU settings, I have changed the configuration part of Ulx3SMinimal.scala.

Like...
```
 //cpu.config.load(VexRiscvConfigs.minimal)
 cpu.config.load(VexRiscvConfigs.muraxLike)
```

I got the following error.

Why is this?
What should I do?

```
[error] Exception in thread "main" spinal.core.SpinalExit:
[error]  Error detected in phase PhaseCheck_noLatchNoOverride
[error] ********************************************************************************
[error] ********************************************************************************
[error] NO DRIVER ON (toplevel/system_cpu_logic_cpu/timerInterrupt : in Bool), defined at
[error]     vexriscv.plugin.CsrPlugin.setup(CsrPlugin.scala:590)
[error]     vexriscv.plugin.CsrPlugin.setup(CsrPlugin.scala:434)
[error]     vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]     vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]     vexriscv.Pipeline$class.build(Pipeline.scala:47)
[error]     vexriscv.VexRiscv.build(VexRiscv.scala:122)
[error]     vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error]     vexriscv.VexRiscvBmbGenerator$$anonfun$7$$anon$2.<init>(VexRiscvBmbGenerator.scala:95)
[error]     vexriscv.VexRiscvBmbGenerator$$anonfun$7.apply(VexRiscvBmbGenerator.scala:90)
[error]     vexriscv.VexRiscvBmbGenerator$$anonfun$7.apply(VexRiscvBmbGenerator.scala:90)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
[error] NO DRIVER ON (toplevel/system_cpu_logic_cpu/externalInterrupt : in Bool), defined at
[error]     vexriscv.plugin.CsrPlugin.setup(CsrPlugin.scala:591)
[error]     vexriscv.plugin.CsrPlugin.setup(CsrPlugin.scala:434)
[error]     vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]     vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]     vexriscv.Pipeline$class.build(Pipeline.scala:47)
[error]     vexriscv.VexRiscv.build(VexRiscv.scala:122)
[error]     vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error]     vexriscv.VexRiscvBmbGenerator$$anonfun$7$$anon$2.<init>(VexRiscvBmbGenerator.scala:95)
[error]     vexriscv.VexRiscvBmbGenerator$$anonfun$7.apply(VexRiscvBmbGenerator.scala:90)
[error]     vexriscv.VexRiscvBmbGenerator$$anonfun$7.apply(VexRiscvBmbGenerator.scala:90)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
```

Dolu1990, 2021-09-30:
you need to manage the interrupts. Do you want them ?

It isn't exactly the same situation, but in the linux soc it is manged as : 
https://github.com/SpinalHDL/SaxonSoc/blob/440ff0eca051e117a47d6e83fb87c6b1e8ec0f1a/hardware/scala/saxon/VexRiscvClusterGenerator.scala#L34

jmio, 2021-09-30:
Thank you .

I don't need external interrupts at the moment, but I do need timer interrupts.

I tried to mimic it as follows, 

```
.....
  implicit val cpu = VexRiscvBmbGenerator()

  interconnect.setDefaultArbitration(BmbInterconnectGenerator.STATIC_PRIORITY)
  interconnect.setPriority(cpu.iBus, 1)
  interconnect.setPriority(cpu.dBus, 2)

  val clint = BmbClintGenerator(0xB00000) //Used as a time reference only
  clint.cpuCount.load(0)

  cpu.setTimerInterrupt(clint.timerInterrupt(0))  // <====== Add This line

  //Add components
  val ramA = BmbOnChipRamGenerator(0x80000000l) // 32K Code
....

```
but got another error (^^;

```
[info] ! SpinalHDL async engine is stuck !
[info] Waiting on system_cpu_timerInterrupt defined at ???:
[info] 1) ??? loader
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 1.154 : Elaborate components
[info] ! SpinalHDL async engine is stuck !
[info] Waiting on system_cpu_timerInterrupt defined at vexriscv.VexRiscvBmbGenerator.<init>(VexRiscvBmbGenerator.scala:34):
[info] 1) ??? loader
[error] Exception in thread "main" java.lang.Exception: SpinalHDL async engine is stuck
[error]         at spinal.core.fiber.EngineContext.start(AsyncCtrl.scala:124)
[error]         at spinal.core.fiber.Engine$.create(AsyncCtrl.scala:150)
[error]         at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:2235)
[error]         at spinal.core.internals.PhaseContext.doPhase(Phase.scala:181)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$120.apply(Phase.scala:2538)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$120.apply(Phase.scala:2536)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2536)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2473)
[error]         at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:32)
[error]         at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2473)
[error]         at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2468)
[error]         at spinal.core.Spinal$.apply(Spinal.scala:371)
[error]         at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:164)
[error]         at saxon.board.muselab.ICESugarPro.ICESugarProMinimal$.delayedEndpoint$saxon$board$muselab$ICESugarPro$ICESugarProMinimal$1(ICESugarProMinimal.scala:213)
[error]         at saxon.board.muselab.ICESugarPro.ICESugarProMinimal$delayedInit$body.apply(ICESugarProMinimal.scala:204)
[error]         at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error]         at scala.App$class.main(App.scala:76)
[error]         at saxon.board.muselab.ICESugarPro.ICESugarProMinimal$.main(ICESugarProMinimal.scala:204)
[error]         at saxon.board.muselab.ICESugarPro.ICESugarProMinimal.main(ICESugarProMinimal.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 10 s, completed 2021/10/01 6:52:55

jmio, 2021-09-30:
Sorry, the error above was my fault.

```
cpu.config.load(VexRiscvConfigs.minimal)
```

jmio, 2021-09-30:
However, I still got the following error  (^^;
‘‘‘
[error] Exception in thread "main" spinal.core.SpinalExit:
[error]  Error detected in phase PhaseNormalizeNodeInputs
[error] ********************************************************************************
[error] ********************************************************************************
[error] Static bool extraction (bit 0) is outside the range (-1 downto 0) of (toplevel/system_clint_logic/io_timerInterrupt : out Bits[0 bits]) at
[error]     spinal.lib.bus.bmb.BmbClintGenerator$$anonfun$timerInterrupt$1.apply(BmbGenerators.scala:95)
[error]     spinal.lib.bus.bmb.BmbClintGenerator$$anonfun$timerInterrupt$1.apply(BmbGenerators.scala:95)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
‘‘‘

Dolu1990, 2021-10-01:
Ho did you instanciated the clint ?

Dolu1990, 2021-10-01:
If you don't want the clint as a timer, then you should not use its io_timerInterrupt 
You can just tie the CPU timer input to False ^^

jmio, 2021-10-02:
I read the definitions of BmbClintGenerator, BmbClint and Clint.
If cpuCount = 0, then no signal line is generated, right?

```
  val clint = BmbClintGenerator(0xB00000) 
  clint.cpuCount.load(1) // must not 0!
```

jmio, 2021-10-02:
It also worked on the actual machine. Thank you.

The behavior of interrupts seems to be different from Briey, so I'll ask again if I don't understand.

Dolu1990, 2021-10-03:
> If cpuCount = 0, then no signal line is generated, right?

Right, 1 timer per cpu ^^ 

> The behavior of interrupts seems to be different from Briey, so I'll ask again if I don't understand.

Yes, the behaviour is very different. SaxonSoc follow the "standard" risc-v way (combo plic / clint)