SpinalHDL/SaxonSoc #68: usb pmod

sjthales, 2022-02-03:
Hello,

Couple of weeks, I saw this video https://www.youtube.com/watch?v=K5Vv32O9g7Q
I would like to run this demo on my side, I have an Arty7 35t, pmod VGA and pmod SDcard.
On the video, I don't see what is the pmod use for USB.
Could you please tell me what is the pmod used?
Regards

Dolu1990, 2022-02-03:
Hi,

I designed it, here are the files : 
https://github.com/Dolu1990/pmod_usb_host_x4

Do what ever you want with it ^^

Regards

sjthales, 2022-02-03:
Thank you for your quick answer.
I don't have the way to make the pmod USB you designed.
Is it possible to use a pmod USB from digilent (by slightly modifying the design) and to connect it to an USB hub?
What do you use 4 USB for?

Dolu1990, 2022-02-03:
There may be out there some existing pmod, maybe some from radiona, but i'm not realy much aware, for sure pin would need to be remapped.

1 for mouse
1 for keyboard
1 for usb drive to boot from
1 spare for eventualy a usb hub and more devices

sjthales, 2022-02-04:
Ok thanks a lot.
I think we will make your usb pmod.