SpinalHDL/SaxonSoc #69: Vexriscv snapshot unresolved dependency path during assembly

eyssartk, 2022-02-03:
Hello,

It seems that a vexriscv SNAPSHOT is missing from https://repo1.maven.org/maven2/default/.
Or maybe my install is faulty ?

```
thales@XPS-15-9560:~/saxonsoc$ sbt assembly
[info] welcome to sbt 1.4.7 (Private Build Java 1.8.0_222)
[info] loading settings for project global-plugins from idea.sbt ...
[info] loading global plugins from /home/thales/.sbt/1.0/plugins
[info] loading settings for project saxonsoc-build from plugins.sbt ...
[info] loading project definition from /home/thales/saxonsoc/project
[info] loading settings for project root from build.sbt ...
[info] set current project to SaxonSoc (in build file:/home/thales/saxonsoc/)
[info] Updating 
[info] Resolved  dependencies
[warn] 
[warn] 	Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading default:vexriscv_2.11:0.1.0-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/thales/.ivy2/localdefault/vexriscv_2.11/0.1.0-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/default/vexriscv_2.11/0.1.0-SNAPSHOT/vexriscv_2.11-0.1.0-SNAPSHOT.pom
[error] 	at lmcoursier.CoursierDependencyResolution.unresolvedWarningOrThrow(CoursierDependencyResolution.scala:258)
[error] 	at lmcoursier.CoursierDependencyResolution.$anonfun$update$38(CoursierDependencyResolution.scala:227)
[error] 	at scala.util.Either$LeftProjection.map(Either.scala:573)
[error] 	at lmcoursier.CoursierDependencyResolution.update(CoursierDependencyResolution.scala:227)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:60)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:53)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:103)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:73)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$20(LibraryManagement.scala:116)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:228)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:116)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:97)
[error] 	at sbt.util.Tracked$.$anonfun$inputChangedW$1(Tracked.scala:219)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:130)
[error] 	at sbt.Classpaths$.$anonfun$updateTask0$5(Defaults.scala:3486)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] 	at sbt.Execute.work(Execute.scala:291)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:282)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:64)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (update) sbt.librarymanagement.ResolveException: Error downloading default:vexriscv_2.11:0.1.0-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/thales/.ivy2/localdefault/vexriscv_2.11/0.1.0-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/default/vexriscv_2.11/0.1.0-SNAPSHOT/vexriscv_2.11-0.1.0-SNAPSHOT.pom
[error] Total time: 2 s, completed 3 févr. 2022 17:40:52
```

tcal-x, 2022-02-03:
Have you loaded the submodules under ext/?    From anywhere in the SaxonSoc repository, run `git submodule update --init`.

eyssartk, 2022-02-03:
Thanks that was it!