SpinalHDL/SaxonSoc #71: SRAM support?

RetroHQ, 2022-03-11:
Could you give any advice on how to implement an SRAM controller?

I see there is the **BmbSramGenerator.scala** in deprecated but this is not compatible with the latest branch and I do not have enough knowledge of any of this to know quite how to migrate it.

Help with this much appreciated.

RetroHQ, 2022-03-11:
After a good amount of comparison between different versions / branches and some good old trial and error I seem to have migrated the blackice SRAM code to something which compiles and generates verilog and valid looking nets.

I may have more queries on this, but it's looking promising and a great way to learn how this HDL translates into verilog -- it's all very much black magic to me atm.

Dolu1990, 2022-03-14:
Hi,

So SaxonSoc was very very experimental in its paradigme in version 1.0, and i guess that BmbSramGenerator is a legacy from that time.

Version 3.0 stabilized things quite a bit, but not everything was ported.

Let me know if something isn't going well.

mixotricha, 2024-01-11:
I have some hardware where I need to implement an sram controller. Wondering if the code referred to above ( or the modified code ) ever got posted anywhere. I tried looking for the depreciated version and could not find.