SpinalHDL/SaxonSoc #72: DMA from peripherals?

RetroHQ, 2022-03-12:
Is it possible to DMA from peripheral to memory? I see there is the DmaApbBmbGenerator commented out, which looks like it would be for this purpose.

in this case I would like the ability to DMA from SPI to memory, which would require DMA trigger on SPI recieve.

Dolu1990, 2022-03-14:
So, that DMA you found was a first implement of a lightweight DMA, but it wasn't pushed all the way in its developpement.
Instead, a second version of the DMA was implement (quite a bit more heavy XD) which handle all the kind of transfers, and also handle linked list (scatter gatther) transfers : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/hardware/scala/saxon/board/digilent/ArtyA7SmpLinux.scala#L77

The DMA itself support peripheral to memory via https://github.com/SpinalHDL/SpinalHDL/blob/e48dc0eb13b3d8139645a81ae824014909f2fb59/lib/src/main/scala/spinal/lib/system/dma/sg/DmaSgGenerator.scala#L115

The DMA itself was fully tested in simulation, then it was deployed on hardware with memory to memory, memory to peripheral transfers. But peripherals to memory wasn't deployed / tested physicaly.

RetroHQ, 2022-03-14:
Ah, great, I did see that but didn’t know if it was what I needed. Are there any code examples of a dma from peripheral to memory?

Dolu1990, 2022-03-16:
There is no example for that direct application. as it was not used in SaxonSoc
For the software  =>  it is close to : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/dmasg/src/main.c#L50
Just it use https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/dmasg.h#L97 instead of dmasg_input_memory