SpinalHDL/SaxonSoc #78: saxon smp linux in verilator - feature request

laurentiuduca, 2022-12-14:
hi,

  i am completely new to spinal hdl and vexriscv
and i have knowledge about verilog linux and buildroot
i compiled Linux.scala and simulated linux in verilator and i like it because it is fast.
i want to simulate dual core with simple uart and SMP linux in verilator
- maybe this can be done with saxon soc or VexRiscvSmpCluster.scala
but i am blocked with system interconnections
i want it simple to be fast and easier to understand.
i think this is an essential feature.

  can you put this feature on your agenda, please
or give some detailed hints

thank you

laurentiuduca, 2022-12-14:
forgot to say that it would be essential 
to have the same interaction with the console as linux.scala offers

laurentiuduca, 2022-12-15:
i have discovered the linux-on-litex-vexriscv repository which does this but has lots of peripherals which slow down the simulation speed

Dolu1990, 2022-12-15:
Personnaly, i have https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/hardware/scala/saxon/board/digilent/ArtyA7SmpLinux.scala#L586

But that's also with many peripherals.

Else, there is a very striped down simulation soc here : 
https://github.com/SpinalHDL/VexRiscv/blob/051d140c33ce1480e10bdf76668fceae8ff59bef/src/main/scala/vexriscv/demo/smp/VexRiscvSmpLitexCluster.scala#L238

It was used to simulate the multicore internals used for Litex.