SpinalHDL/SaxonSoc #79: Support for multiple SoC

surabibio, 2023-01-15:
I would like to use 2 SoC. (e.g one exclusively for WiFi and another exclusively for Data Acquisition) I am able to get 2 Verilog files but am unable to synthesis them due to **Verilog Error: (VERI-1206) overwriting previous definition of module**. 
The work around is rename the duplicate modules (e.g module _VexRiscv_ to _wifi_VexRiscv_ and _daq_VexRiscv_  ...) Is there a way to generate modules with unique names.

Dolu1990, 2023-01-16:
Hi,

There is mainly two way to solve this : 
- Either you incluse each soc in a different Verilog package in your synthesis tool
- Either you use the feature which add a global prefix on everything : 
- 
```scala
SpinalRtlConfig.copy(globalPrefix = "soc1").generateVerilog(...)
SpinalRtlConfig.copy(globalPrefix = "soc2").generateVerilog(...)
```

surabibio, 2023-01-17:
Hi Charles,
Wondering how I missed "globalPrefix" and googling for the same, I came across your other solution:
`SpinalConfig(privateNamespace = true)`
This is very elegant. I used the same and now dual SoC is working great.
Thanks
Ravi