SpinalHDL/SaxonSoc #81: ArtyA7SmpLinux saxon_netlist faild!

chshux, 2023-12-27:
Hi,
    When I execute Saxon_netlist on the terminal, the command was failed. Below is the log that was run:

chenshux@chenshux:~/ic-projs/ArtyA7SmpLinux/SaxonSoc$ saxon_netlist
bsp        cpu1.yaml        hardware         makefile   software
build.sbt  ext              LICENSE          project    target
cpu0.yaml  formatsdcard.sh  linux2sdcard.sh  README.md  test
[info] welcome to sbt 1.4.7 (Private Build Java 1.8.0_392)
[info] loading settings for project saxonsoc-build from plugins.sbt ...
[info] loading project definition from /home/chenshux/ic-projs/ArtyA7SmpLinux/SaxonSoc/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project vexriscv-build from plugins.sbt ...
[info] loading project definition from /home/chenshux/ic-projs/ArtyA7SmpLinux/SaxonSoc/ext/VexRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /home/chenshux/ic-projs/ArtyA7SmpLinux/SaxonSoc/ext/SpinalHDL/project
[info] loading settings for project all from build.sbt ...
[info] set current project to SaxonSoc (in build file:/home/chenshux/ic-projs/ArtyA7SmpLinux/SaxonSoc/)
[info] running (fork) saxon.board.digilent.ArtyA7SmpLinux 
[info] [Runtime] SpinalHDL dev    git head : 4fd59f09e99b2a3a9e2d5fa00fdad9a536652ae2
[info] [Runtime] JVM max memory : 3492.0MiB
[info] [Runtime] Current date : 2023.12.27 09:51:06
[info] [Progress] at 0.000 : Elaborate components
[info] cpuDecode to fpuDispatch 1
[info] fpuDispatch to cpuRsp    2
[info] cpuWriteback to fpuAdd   1
[info] add                      4
[info] mul                      5
[info] fma                      10
[info] short                    2
[info] [Progress] at 3.478 : Checks and transforms
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (21 errors).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 3.828 : Elaborate components
[info] cpuDecode to fpuDispatch 1
[info] fpuDispatch to cpuRsp    2
[info] cpuWriteback to fpuAdd   1
[info] add                      4
[info] mul                      5
[info] fma                      10
[info] short                    2
[info] [Progress] at 4.995 : Checks and transforms
[error] Exception in thread "main" spinal.core.SpinalExit: 
[error]  Error detected in phase PhaseAnalog
[error] ********************************************************************************
[error] ********************************************************************************
[error] Assignment data type mismatch
[error] (toplevel/sdramDomain_phyA_logic_phy/??? :  Bool) := (toplevel/sdramDomain_phyA_logic_phy/io_sdram_DQ : inout Bits[16 bits])
[error]     saxon.board.digilent.ArtyA7SmpLinux$.main(ArtyA7SmpLinux.scala:453)
[error]     saxon.board.digilent.ArtyA7SmpLinux.main(ArtyA7SmpLinux.scala)
[error] ********************************************************************************
[error] ********************************************************************************
[error] Assignment data type mismatch
[error] (toplevel/sdramDomain_phyA_logic_phy/??? :  Bool) := (toplevel/sdramDomain_phyA_logic_phy/io_sdram_DQ : inout Bits[16 bits])
[error]     saxon.board.digilent.ArtyA7SmpLinux$.main(ArtyA7SmpLinux.scala:453)
[error]     saxon.board.digilent.ArtyA7SmpLinux.main(ArtyA7SmpLinux.scala)
[error] ********************************************************************************
[error] ********************************************************************************
[error] Assignment data type mismatch
[error] (toplevel/sdramDomain_phyA_logic_phy/??? :  Bool) := (toplevel/sdramDomain_phyA_logic_phy/io_sdram_DQ : inout Bits[16 bits])
[error]     saxon.board.digilent.ArtyA7SmpLinux$.main(ArtyA7SmpLinux.scala:453)
[error]     saxon.board.digilent.ArtyA7SmpLinux.main(ArtyA7SmpLinux.scala)
[error] ********************************************************************************
[error] ********************************************************************************
[error] Assignment data type mismatch
[error] (toplevel/sdramDomain_phyA_logic_phy/??? :  Bool) := (toplevel/sdramDomain_phyA_logic_phy/io_sdram_DQ : inout Bits[16 bits])
[error]     saxon.board.digilent.ArtyA7SmpLinux$.main(ArtyA7SmpLinux.scala:453)
[error]     saxon.board.digilent.ArtyA7SmpLinux.main(ArtyA7SmpLinux.scala)
[... 98 lines cut ...]
[error] Design's errors are listed above.
[error] SpinalHDL compiler exit stack : 
[error] 	at spinal.core.SpinalExit$.apply(Misc.scala:440)
[error] 	at spinal.core.SpinalError$.apply(Misc.scala:495)
[error] 	at spinal.core.internals.PhaseContext.checkPendingErrors(Phase.scala:174)
[error] 	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:190)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$141.apply(Phase.scala:2746)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$141.apply(Phase.scala:2744)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2744)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2680)
[error] 	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:69)
[error] 	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2680)
[error] 	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2675)
[error] 	at spinal.core.Spinal$.apply(Spinal.scala:405)
[error] 	at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:173)
[error] 	at saxon.board.digilent.ArtyA7SmpLinux$.main(ArtyA7SmpLinux.scala:453)
[error] 	at saxon.board.digilent.ArtyA7SmpLinux.main(ArtyA7SmpLinux.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 7 s, completed 2023-12-27 9:51:12

I followed the steps in [README](https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/bsp/digilent/ArtyA7SmpLinux/README.md). 
The running environment is Ubuntu 22.

Dolu1990, 2024-01-04:
Hi,

I update the dev-0.3 branch with upstream SpinalHDL VexRiscv, it now seems good on my side.

Let's me know if you have any issue ^^

Note i'm now working on VexiiRiscv, will be based on tilelink interconnect.

Dolu1990, 2024-04-04:
Hi,

Thanks for the notification, the vexriscv in saxonsoc was a bit behind and had some unwanted push in it XD

Fixed now

commit 0110da76 (2024-04-04): Fix #81
closed 2024-04-04 by Dolu1990